JoeyDelp/JoSIM #18: Some issues about building this project

tsyw, 2018-07-09:
Environment: Win10+Visual Studio Community 2017 with the basic C++ for desktop installed

Here is my building result:
It says that it cannot find the file  j_model.cpp.
Does this file exist? Or is this file user defined?

One more question, will the conversion warning have any influence on the usage? 

Sorry for the disturbance.

1>------ Rebuild All started: Project: JoSIM, Configuration: DEBUG_None x64 ------
1>j_components.cpp
1>j_errors.cpp
1>j_input.cpp
1>josim.cpp
1>j_matrix.cpp
1>c:\users\tsyw\desktop\josim\josim-master\src\j_matrix.cpp(1353): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_matrix.cpp(1354): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>j_misc.cpp
1>c:\users\tsyw\desktop\josim\josim-master\src\j_misc.cpp(292): warning C4244: 'argument': conversion from 'double' to 'const unsigned __int64', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_misc.cpp(334): warning C4244: 'argument': conversion from 'double' to '__int64', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_misc.cpp(342): warning C4244: 'initializing': conversion from 'double' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_misc.cpp(376): warning C4244: 'initializing': conversion from 'double' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_misc.cpp(377): warning C4244: 'initializing': conversion from 'double' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_misc.cpp(411): warning C4244: 'initializing': conversion from 'double' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_misc.cpp(482): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data
1>j_models.cpp
1>c1xx : fatal error C1083: Cannot open source file: '..\j_models.cpp': No such file or directory
1>j_output.cpp
1>j_parser.cpp
1>c:\users\tsyw\desktop\josim\josim-master\src\j_parser.cpp(27): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_parser.cpp(32): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>j_plot.cpp
1>c:\users\tsyw\desktop\josim\josim-master\src\j_plot.cpp(485): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_plot.cpp(487): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_plot.cpp(489): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>j_simulation.cpp
1>c:\users\tsyw\desktop\josim\josim-master\src\j_simulation.cpp(78): warning C4244: 'initializing': conversion from 'double' to 'int', possible loss of data
1>c:\users\tsyw\desktop\josim\josim-master\src\j_simulation.cpp(235): warning C4244: '=': conversion from 'double' to 'int', possible loss of data
1>Generating Code...
1>Done building project "josim.vcxproj" -- FAILED.
========== Rebuild All: 0 succeeded, 1 failed, 0 skipped ==========

tsyw, 2018-07-09:
Also, if possible, can you offer a checklist of to-do list?
Thank you again for your contribution.

tsyw, 2018-07-09:
Ah, I saw you commit logs, you change this file all to jj_input.cpp. 
So I have another question, is the mtj model, ntron model you offered usable and how can I define my own model?

JoeyDelp, 2018-07-12:
Good day, sorry for the late reply, I have been out of the office.
Yes indeed the jj_model.cpp has been entirely scrapped and the classes moved to the jj_input.cpp file, I am working on an update to the VS solution and will push it as soon as it is done.

The conversions should not have any effect on the the usage, they will be ironed out with time as data structures are improved.

The MTJ and Ntron models are still a WIP with no expected ETA at this time. I will however update them as soon as they are usable.

closed 2018-07-12 by JoeyDelp