JoeyDelp/JoSIM #19: Windows: How to change the input path file?

Hassan313, 2018-10-10:
Dear Joey,

Thank you for the tool. I want to know should we change the josim.vcxproj.user file to change the input path file? If yes, we should reopen the visual studio and reopen the solution to change the input file to the simulator?

Thank you.

Best Regards
Hassan

JoeyDelp, 2018-10-10:
Good morning,
The josim.vcxproj.user is project file which only contains the paths to files used when testing in debug and release builds in Visual Studio on Windows. The simulator alone does not require these vcxproj files to work. It is a command line utility which can be run using the instructions in the user manual. 

If you are compiling the project on Windows and wish to test the compiled program then yes it would be wise to alter the paths to the files in the project configuration within Visual Studio itself rather than the vcxproj.user file. 

Hope this answers your question.
Best regards,
Joey

Hassan313, 2018-10-10:
Dear Joey,

Thank you for you response. Now I have another problem which is the Anaconda 3 Python 3.7. When I try to compile it is give me the below error.

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1181	cannot open input file 'python36.lib'	JoSIM	C:\\JoSIM-master\src\josim_vs\LINK	1	
 Can you kindly advise how to fix this issue?

Thank you.

Best Regards
Hassan

closed 2018-10-14 by JoeyDelp