JoeyDelp/JoSIM #21: Anaconda 3 Python Version 3.7 problem

Hassan313, 2018-10-10:
Dear Joey,

I have a problem which is the Anaconda 3 Python 3.7. When I try to compile it is give me the below error.

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1181	cannot open input file 'python36.lib'	JoSIM	C:\JoSIM-master\src\josim_vs\LINK	1
Can you kindly advise how to fix this issue?

Thank you.

Best Regards
Hassan

Hassan313, 2018-10-10:
Even if I go with the Python 3.6 version, I get the following error message:
[image]

JoeyDelp, 2018-10-11:
Good afternoon,

Did you set the environment variables as stated under the Windows 10 section of the[ReadMe.md](https://github.com/JoeyDelp/JoSIM/blob/master/README.md)?

These environment variables in Windows tell the system where to find the python36.lib.

Hope this helps

Hassan313, 2018-10-11:
Dear Joey,

Yes. I did. 

Is your software compatible with the **Python 3.7**?

Thank you.

Best Regards
Hassan

JoeyDelp, 2018-10-12:
Good day, it should be, simply edit the Linker settings inside project settings of Visual Studio under Input->Additional Dependencies.

Change python36.lib to python37.lib which should fix the issue. Once compiled it should work on any python version that is installed on the system.

Kind regards

Hassan313, 2018-10-12:
Dear Joey, 

Good Evening,

Thank you for the information.

Best Regards
Hassan

closed 2018-10-14 by JoeyDelp