JoeyDelp/JoSIM #22: Python 3.6 DLL Files Missing

Hassan313, 2018-10-10:
Dear Joey, 

When I tried the Python 3.6 point, the below problem appeared: 
[image]

Hassan313, 2018-10-10:
I downloaded the Python 3.6 DLL file from the below website 

https://www.dll-files.com/python36.dll.html

and added the DLL file to the bin folder where the executable files are, the above problem disappeared, however the below problem appeared:

[image]

It is saying that "ImportError: numpy.core.multiarray failed to import".

Please kindly help me with this issue.

Best Regards
Hassan

Hassan313, 2018-10-11:
Dear Joey,

Thank you for your response.

I have followed as you have said in this email. It did not work. The problem was I had to add the Anaconda3 directory to the path variable of system.

Thank you.

Best Regards
Hassan

JoeyDelp, 2018-10-12:
My apologies, it might be necessary to add the %PYTHONHOME% to the PATH variable as well. This however has not been necessary on some systems. I will update the readme to include this.

Hassan313, 2018-10-12:
Dear Joey,

Thank you.

Best Regards
Hassan

closed 2018-10-14 by JoeyDelp