JoeyDelp/JoSIM #55: Error while building the JoSIM

anadb, 2021-11-14:
Hi!

I am trying to install JoSIM but I have an error

`In file included from /home/anatoly/Documents/JoSIM/include/JoSIM/BasicComponent.hpp:6:0,
                 from /home/anatoly/Documents/JoSIM/include/JoSIM/Capacitor.hpp:6,
                 from /home/anatoly/Documents/JoSIM/src/Capacitor.cpp:4:
/home/anatoly/Documents/JoSIM/include/JoSIM/TypeDefines.hpp:8:20: fatal error: optional: No such file or directory
compilation terminated.
CMakeFiles/josim.dir/build.make:75: recipe for target 'CMakeFiles/josim.dir/src/Capacitor.cpp.o' failed
make[2]: *** [CMakeFiles/josim.dir/src/Capacitor.cpp.o] Error 1
CMakeFiles/Makefile2:293: recipe for target 'CMakeFiles/josim.dir/all' failed
make[1]: *** [CMakeFiles/josim.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
`

JoeyDelp, 2021-11-17:
Hi, please let me know what version of Linux Mint as well as which C++ compiler and version you are using.

anadb, 2021-11-17:
thank you for the answer

I have linux mint version 18.3 (Sylvia) and g++ 5.40. However I can compile my C++17 programs with option -std=c++17

JoeyDelp, 2021-11-17:
Thanks, let me try and replicate the issue on my side and see if I can find a solution to the issue.

JoeyDelp, 2021-11-17:
I was able to reproduce the error in a Linux Mint 18.3 VM on my side. The issue is definitely compiler related with features not available in older compilers. I was able to quite efficiently and unobtrusively update GCC to version 8 using a modified version of the following gist: https://gist.github.com/youmych/122e32cfa1c228c84684fec13b7d52e1#file-how-to-install-latest-gcc-on-linux-mint-txt

sudo apt-get update && \
sudo apt-get install build-essential software-properties-common -y && \
sudo add-apt-repository ppa:ubuntu-toolchain-r/test -y && \
sudo apt-get update && \
sudo apt-get install gcc-snapshot -y && \
sudo apt-get update && \
sudo apt-get install gcc-8 g++-8 gcc-multilib -y && \
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 80 --slave /usr/bin/g++ g++ /usr/bin/g++-8

You can always change back to GCC 5 by simply running:
sudo update-alternatives --config gcc

Using this method I was successfully able to compile JoSIM with all tests passing succesfully. 

Let me know this solves the issue.

closed 2021-11-17 by anadb