JoeyDelp/JoSIM #66: Usage

ToM-ynu, 2022-03-11:


JoeyDelp, 2022-03-15:
Hi,

I found some issues with the custom waveform function. There was some changes to the functions going from 2.4 to 2.5 and now 2.6 which seem to have affected the custom waveform but I did not pick up on. 

I fixed the issues and it should work now for all interpolation methods. Attached is the output of the custom waveform in your example. Please have a look and see if the version under testing branch solves the issues you have. If they do I will pull this as a fix into the master branch.

Best regards,
Joey
[image]

ToM-ynu, 2022-03-16:
We built the source code in the testing branch and verified that the cus power supply is working properly. Thank you so much for your quick response.

In our environment, due to the effect of capillarizing the netlist, we are not able to specify the file correctly. By Placing the CUS.TXT in the path written in all capitals solved the problem. Since this is not a functional problem, could you please change it to prevent capillarization of file names when you have time?

Translated with www.DeepL.com/Translator (free version)

JoeyDelp, 2022-03-16:
Good morning,

I have pushed another fix to the  testing branch that should check both `CUS.TXT` (as created by JoSIM internally due to the toupper function) and if it does not find this will look for the lowercase `cus.txt`. 

Please note that this prevents any form of CaMeL cAsE file names. The txt file needs to be either fully lower case or fully upper case.

Let me know if this works.

Best regards,
Joey

ToM-ynu, 2022-03-17:
Hello.

I confirmed it is works!
Thank you very much for your help.

closed 2022-03-23 by JoeyDelp