JoeyDelp/JoSIM #81: Subcircuit questions

grahamrow, 2022-10-19:
Hi all, hope you're doing well. Was testing Josim (like the cmake build system!) and found some issues with subcircuits regarding input nodes labels and parameters.

1. Subcircuit input Node labels cannot be referenced in `.print` statements, e.g.:
```spice
X1 TEST 1 0
I1      0 1 1.0

.subckt TEST IN GND
    R1 IN GND 10.0
.ends

.tran 0.25p 5n 0 1.0p
.print v(IN.X1)
.end
```
yields
```
W: Controls
Unknown device/node IN|X1
Cannot store results for this device/node.
Ignoring this store request.
```
Expected behavior would be to look up the proper node label based on the input mapping. To be fair this is easy to work around, but is expected behavior for those coming from WRspice (IIRC).
2. Subcircuit parameters appear to be broken (unless I'm the broken one!), e.g.:
```
X1 TEST 1 0 RES=10.0
I1      0 1 1.0

.subckt TEST IN GND
    R1 IN GND RES
.ends

.tran 0.25p 5n 0 1.0p
.print v(1)
.end
```
yields 
```
E: Parsing
The following variables were not found/defined.
Please ensure that these variables exist within the netlist.
Variables: 
RES
```
Another attempt, trying to interpret the manual's instructions for declarations:
```
X1 TEST 1 0 RES=5.0
I1      0 1 1.0

.subckt TEST IN GND RES=10
    R1 IN GND RES
.ends

.tran 0.25p 5n 0 1.0p
.print v(1)
.end
```
which yields
```
E: Input
The IO of line "X1" does not match the subcircuit IO.
Please check the line and ensure correct IO and that parameters do not contain spaces.
```
Expected behavior is to substitute the indicated LABEL=VALUE pairs into the subcircuit instantiation, as indicated in the manual and by custom. I'll try to dig through the source a bit, but wanted to bring this up. Thanks for pushing this project forwards!

grahamrow, 2022-10-19:
Okay, in ``void Netlist::expand_io`` of Netlist.cpp the code only checks the first token in the line for a match to the parameters:
```c++
    if (params.count(tokens.front()) != 0) {
       std::cout << "-> Found a match ";
       insert_parameter(tokens, params);
    }
```
where really it should iterate over each token, i.e.:
```c++
    int64_t match_count = 0;
    for (auto & element : tokens) {
      if (params.count(element) != 0) {
          match_count++;
      }
    }
    if (match_count != 0) {
      insert_parameter(tokens, params);
    }
```
Though this creates an `E: Out of range error. This is a bug` so clearly some more tweaks are needed...

grahamrow, 2022-10-20:
Okay, I think I have this solved, will open a pull request since continuing to comment will surely be annoying for everyone 😄

JoeyDelp, 2022-10-21:
Hi Graham, thanks for raising this issue. 

The first issue is one that I can easily add to the functionality as you mentioned, since it's just a matter of keeping track of which subcircuit node maps to which internal node when expanding the netlist. 

The second I will try to address more clearly in the manual. The first example you tried was almost correct, and the error message kind of hints to the problem. What was missing was the `.param RES=<some default>` statement in the subcircuit.

```
X1 TEST 1 0 RES=10.0
I1      0 1 1.0

.subckt TEST IN GND
    .param RES=5.0
    R1 IN GND RES
.ends

.tran 0.25p 5n 0 1.0p
.print v(1)
.end
```

Without the `.param` statement, the simulator doesn't know that the `RES` parameter exists and cannot alter it when instantiating the `X1` line. I could make it so that the `RES=10.0` part of the `X1` statement creates a param called `RES` for the `TEST` subcircuit, however, then if omitted there is no default value for the `RES` parameter.

grahamrow, 2022-10-27:
Thanks Joey,

Ahh, okay interesting that works well for RLCs, etc.. I'd been planning to use parameters for amplitudes and frequencies on sources with expression-based source terms. How would that work if there's no explicit association with the parameter name? e.g.:
```
X1 TEST 1 0 AMP=1.0 FREQ=1e9 (can't use I1 here since there are multiple parameters)
I1      0 1 1.0

.subckt TEST IN GND
    R1 IN GND 10.0
    I1 GND IN sin(0 AMP FREQ)
.ends

.tran 0.25p 5n 0 1.0p
.print v(1)
.end
```
I'd messed around with just searching all tokens in all subcircuit lines and replacing any token with a match based on the parameter/value map, but the tokens get parsed with characters like ")" and "(" and presumably some other arithmetical operators so I wanted to pause and check with you.

brainkz, 2022-12-12:
Hi @grahamrow and @JoeyDelp ,

I use JoSIM a lot in my research and wrote a [utility](https://github.com/brainkz/Netlist-Flattener/blob/main/README.md) to flatten the hierarchical netlist while considering named parameters. The flattened netlist is then passed to `josim-cli`. 

It's been working well for me and I added some extra tweaks to make it work with the tests in JoSIM. Perhaps, this script could be polished and integrated into JoSIM, if that is of any interest?

JoeyDelp, 2022-12-13:
Hi @brainkz, this functionality is already present in JoSIM. When you run with verbose level 3 (`-V 3') a flattened netlist with named parameters substituted is produced on screen.

brainkz, 2022-12-13:
Thanks for the heads up, and sorry that I did not write my message clearly. The purpose of the utility extends JoSIM's support to named parameters, thus enabling definitions with keyword parameters
```
...
.subckt COMPARATOR in out clk area_top=2.0 area_bot=2.5 lout=1e-12
    Btop clk in  jjmit area=area_top
    Bbot in  0   jjmit area=area_bot
    Lout in  out lout
.ends
X1 COMPARATOR a b t area_top=4 area_bot=5 lout=1e-11
X2 COMPARATOR c d t area_top=3
...
```
to yield a netlist where parameters `area_top=4 area_bot=5 lout=1e-11` and `area_top=3` in, respectively, `X1` and `X2` are propagated into subcircuit (akin to HSPICE)
```
...
BTOP|1 T A JJMIT AREA=4
BBOT|1 A 0 JJMIT AREA=5
LOUT|1 A B  1e-11
BTOP|2 T C JJMIT AREA=3
BBOT|2 C 0 JJMIT AREA=2.5
LOUT|2 C D  1e-12
...
```
(Node renaming was inspired by `Netlist.cpp`)

JoeyDelp, 2026-11-07:
Both parts are fixed in 3.0.0.

`.print V(IN.X1)` now follows the subcircuit pin mapping. Your example returns 10 V in both analysis modes, and invalid trace requests produce an error instead of being dropped.

Named parameters are now passed to each instance, with header defaults used where no override is given. This includes values used in source expressions and junction parameters. Your RES=10 and default-10/override-5 examples return 10 V and 5 V respectively in both modes.

The [subcircuit documentation](https://joeydelp.github.io/JoSIM/syntax/#subcircuits) describes the binding rules. I'm closing both reported problems as fixed.

closed 2026-11-07 by JoeyDelp