JoeyDelp/JoSIM #82: The issues about ivcurve.py

ZhangLujiu, 2022-12-04:
Hello Joey,

When I use the ivcurve.py file, I need to supply it with two command-line arguments, but I don't know what those two arguments should be. As shown in the figure, what do argv[1] and argv[2] refer to? Did you tell me how to use this file correctly?
[image]

In addition to, I hope you can provide more examples for beginners and  explain in more detail what each line in XXX.cir does or how it should be set up.

Thank you.

Best Regards

Peng Zhang

JoeyDelp, 2022-12-05:
Hi Peng,

The first argument is the simulator (path to the executable) and the second argument is the file used to generate the IV curve. This script was just a quick way I used to generate IV curves before I implemented the functionality inside JoSIM itself (https://joeydelp.github.io/JoSIM/syntax/#iv-curve). This is a more reliable way of producing IV curves. 

If you wish to continue using the IV curve script, then just point the first argument to the josim-cli executable and the second argument to a file with the following content:
```
* Example to simulate an IV curve in JoSIM
IS         0          1          pwl(0      0 10p 0 50p 25E-7 )
B1         1          0          jj1        area=1
*R1         1          0          8        
.model jj1 jj(rtype=0, vg=2.8mV, cap=0.07pF, icrit=0.1mA, rn=16, r0=160)
.tran 0.05p 1n 0 0.1p
*.iv jj1 200u iv1.csv
.print DEVI IS
.print NODEV 1 0
.print PHASE B1
.end
```

I include the (commented) .iv command to show how it can be done with JoSIM itself.

The most basic example (of a JTL) is found here https://joeydelp.github.io/JoSIM/ex_usage/ with each line is quite well explained in https://joeydelp.github.io/JoSIM/syntax/

There are quite a few other examples in the test directory of the source code. Let me know if you have any issues understanding any of the commands in the syntax guide and I will try to explain it better.

Best regards

Joey

ZhangLujiu, 2022-12-05:
Joey, thank you for your careful explanation.  Unfortunately, I didn't succeed in drawing the I-V curve. I will show you my operation process and  input file. Could you tell me which step caused the problem?
`* Josephson junction component test
* Date modified: 2020/01/22
B1  1   0  jj1   area=1
Itest   0   1   pwl(0 0 10p 0 50p 250u 200p 250u 240p 0 250p 0 300p -250u 450p -250u 500p 0)
.model jj1 jj(rtype=1, vg=2.8mV, cap=0.07pF, r0=160, rN=16, icrit=0.1mA)
.tran 0.01p 500p
.print devv B1
.print devi B1
.print devp B1`
The file I used is the example file you provided, and its name is jj.cir,  then I execute "josim-cli  ./josim-cli  jj.cir. After running, the following error will be reported
[image]

But this error will not occur when I use the input file in your reply. Could you please point out the where the error comes from?

I'm sorry to ask you such a detailed question, because I'm a beginner and unfamiliar with Josephon junction simulation, so I don't know exactly what each Label in the.cir file means. Do you have any suggestions? How can I familiarize myself with this knowledge?

ZhangLujiu, 2022-12-19:
Thank you for your answe.  I have a new question about the time step setting. It is that how to determine the  value of T_{STEP}  in the line .tran T_{STEP}? To be specific, when we apply a specified voltage (V) to the Josephson junction then we get a supercurrent of AC and the period of this supercurrent is T=h/(2eV). Does the value of time step take into account the period of AC supercurrent?

JoeyDelp, 2023-01-03:
Hi, my apologies for the late response. The T_{STEP} will automatically take into account the AC period by reducing the size the user specified as T_{STEP} when an error is detected (i.e. the phase of a junction changes too rapidly between steps). This way the simulator will always provide an accurate result during junction switching no matter the size the user specifies in the T_{STEP}. It will, however, honor the user specified value during output if the user did not specify a print step P_{STEP} by only storing the result at every T_{STEP}

JoeyDelp, 2023-03-22:
Hi Mengei-Zhao,

JoSIM cannot do this with the internal routine. However, using the `ivcurve.py` script with a netlist that represents this curve but measuring the voltage at node 1 (between L1 and L2) would allow you to obtain the IV curve of this circuit.

Hope this helps.

Mengfei-Zhao, 2023-03-23:
Hi Joey,

Thank you for reply. I obtained the iv curve by writing a batch program based on pwl of josim.

Mengfei-Zhao, 2023-06-17:
@JoeyDelp Hi Joey, 
I have a question. Is josim a fixed-step simulation or a variable-step simulation. I didn't find a description of this in the documentation. Sorry for asking under the wrong issuse. 

Looking forward to your reply.

Mengfei-Zhao, 2023-06-20:
Hi Joey,

I got it. Thank you for reply. 😄

Mengfei-Zhao, 2023-06-25:
@JoeyDelp Hi Joey,
I have a question. Is JoSIM using the Newton's iteration method, or the modified fixed point method as JSIM? or something else? 😃 

Looking forward to your reply.

JoeyDelp, 2026-11-07:
I've updated the [IV/sweep documentation](https://joeydelp.github.io/JoSIM/sweeps/) and [beginner example](https://joeydelp.github.io/JoSIM/ex_usage/) for 3.0.0. To answer the questions accumulated in this thread:

- `ivcurve.py` has been retired. It could overwrite temporary files and reuse old output after a failed simulation. Use `.IV` for a single-junction model curve, or `tools/circuit_sweep.py` for a complete circuit. The guide includes a SQUID example.
- Restarting the circuit at each bias point and running one continuous PWL sweep give different state histories, so the curves need not coincide. `.IV` now accepts `TSTEP`, `TSTOP` and `TSTART` to control its timestep, duration and averaging window. The default short run does not guarantee a settled DC curve.
- JoSIM uses BDF2, predicts the nonlinear junction terms from previous samples and performs one sparse linear solve per step. It does not iterate Newton within each step or provide SPICE RELTOL/ABSTOL controls.
- Each attempt uses a fixed timestep. If the junction phase advances too quickly, JoSIM can restart the whole run at half that step. My earlier statement that this always guarantees an accurate result was incorrect. Timestep, settling and averaging-window convergence still need to be checked.

I'm closing this now that these questions are answered and the unsafe script has been replaced.

closed 2026-11-07 by JoeyDelp