JoeyDelp/JoSIM #84: Parameter sweep simulations

UGreco, 2023-04-06:
Hi Joey,
A big thank you for creating JoSim!
I would like to ask you if it is possible in JoSim to perform simulations sweeping a parameter automatically.
In the documentation I see that with the .param command you can define a variable X and its value, but I don't understand if you can change the value of X in a range.

For example, I want to perform five transient analysis where I change the value of a resistor Rx each time in the range Rx = (1,2,3,4,5).
Can I do that using the .param command?
Best

Angelo

JoeyDelp, 2023-04-13:
Hi Angelo,

Unfortunately this functionality is not supported in JoSIM. It gets quite complicated when multiple simulations need to be run with different parameters, especially in storing these parameters. Resetting the entire simulation internally to change 1 component would be fine but if there are multiple then every possible combination needs to be simulated and written as output in some way. 

I.e. if you had Rx = (1,2,3,4,5) and Rz = (1,2,3,4,5) then the total combinations would be 5^2, however introducing more parameters to sweep could lead to N^N combinations that all need to be kept track of and output as different files. This could become very unwieldy.

We rather leave this to the user to do with an external script or program. It is for this reason that the library interface libjosim and python module pyjosim exists. 

Hope this helps,
Joey

UGreco, 2023-04-27:
Hi Joey,
Thank you very much for your reply.
Best regards

Angelo

JoeyDelp, 2026-11-07:
3.0.0 now includes `tools/circuit_sweep.py`, which runs the R=1,2,3,4,5 example automatically. It starts a fresh simulator process for each value, keeps the input template unchanged and puts each result in its own directory. The example was checked against V=IR in both analysis modes.

`.PARAM` still sets a value for one run; it does not accept a range. The [sweep guide](https://joeydelp.github.io/JoSIM/sweeps/#five-independent-resistance-points) gives the template and command for doing this with the helper.

I'm closing this since the requested sweep now has a supplied, tested workflow.

closed 2026-11-07 by JoeyDelp