JoeyDelp/JoSIM #90: Parameter expression in PWL time value or .tran command?

camronblackburn, 2023-08-09:
Thanks for making JoSIM - it's great and has been super straight forward to get up to speed with!

However, I'm a bit confused on a "Cannot convert string to double" error I keep getting when using the .param command. I run into the issue when trying to define a .param expression for variables passed into PWL time points or .tran analysis tstep or tstop values. I've played around with the format that I'm using to define the parameter expression (i.e. "2m" vs "2e-3" vs "0.002" etc.), but nothing seems to work. Is it just not possible to define variables for these inputs? 

Thanks!

JoeyDelp, 2023-08-10:
Hi Camron, 

There was a specific reason why we did not enable parameterized time values but it escapes me as to what that was. Might have been during early development and overlooked as improvements were made.

None the less, I fixed it so that these values are also parsed by the parameter parser instead of simply running through the double conversion. 

I added a `ex_jtl_param_times.cir` as an example to verify that it works but also to demonstrate how this can be used.

Hope this helps. I will put out a release soon. It's only on the `master` branch for now if you wish to compile your own version.

JoeyDelp, 2023-08-10:
Latest release has been published with this change. It is v2.6.7

camronblackburn, 2023-08-10:
Awesome - thanks so much for the quick change!! I'll try it out.

closed 2025-07-28 by JoeyDelp