JoeyDelp/JoSIM #93: Allow specifying the Stewart-McCumber parameter in JJ models

dolphingarlic, 2024-04-14:
When working with overdamped Josephson junctions, it's often convenient to specify them in terms of their Stewart-McCumber parameter $\beta_c$ instead of their shunt resistance(s). Would it be possible to allow us to specify $\beta_c$ when declaring a model? $\beta_c$ is related to the other model parameters as:
$$\beta_c = \frac{2\pi C R^2 I_c}{\Phi_0}$$

JoeyDelp, 2026-11-07:
I'm leaving this open for direct beta_c input in the model. 3.0.0 does not add that field.

You can now use the documented parameter expression `R = sqrt(beta_c*Phi0/(2*pi*Ic*C))` for positive beta_c, Ic and C with a constant shunt. The [example](https://joeydelp.github.io/JoSIM/model_limits/#beta-c) has been checked for beta_c=0.2, 1 and 5 in both modes.

For a direct model field, the resistance convention still needs to be defined. RTYPE=0 uses RN, while RTYPE=1 has different subgap, transition and normal-state conductances. Choosing R0 or RN gives different damping assumptions.

JoeyDelp, 2026-11-07:
Implemented in [JoSIM 3.1.0](https://github.com/JoeyDelp/JoSIM/releases/tag/v3.1.0).

For a constant-resistance junction, specify the nominal Stewart–McCumber parameter directly:

```spice
.model JM JJ(RTYPE=0 IC=100u C=1p BETA_C=1)
```

JoSIM calculates RN from BETA_C, Ic and C before applying instance scaling. All three must be finite and positive. BETA_C requires RTYPE=0 and cannot be combined with explicit RN or temperature-dependent model parameters.

The [damping guide](https://joeydelp.github.io/JoSIM/model_limits/#beta-c) gives the equation and examples. Regression tests cover the conversion and junction response in both analysis modes.

closed 2026-11-07 by JoeyDelp