JoeyDelp/JoSIM #94: Missing energy in a filter simulation

alicenprzemek, 2024-07-24:
Hello,
I am trying to simulate frequency response of a filter with JoSIM by creating a Gaussian excitation and then doing Fourier analysis on the output. I saw that the S-parameters I calculate don't add up to 1 so I did the same time-domain based analysis using qucs-s and I can't get them matching. The results from qucs-s fit S-parameter simulation. 

Here is a plot comparing the voltage at resistor P2:
[image]

And the power balance from JoSIM:
[image]
(where we expect a flat line at 1)

I feel I did double check my calculations but nonetheless can't find where could be a problem. Therefore it is not clear to me if I am using JoSIM correctly?

```
CC1 nodeinput midpoint 50f
LL1 nodeoutput midpoint 10n
LL2 nodeinput gnd 1n
LL3 nodeoutput gnd 1n
VIN nodeinput nodeinputR cus(waveform.txt 10p 1.0 2)
RP1 gnd nodeinputR  50
RP2 gnd nodeoutput  50
.tran 10p 25000p 
.print NODEV nodeinput
.print DEVI RP1
.print NODEV nodeoutput
.print DEVI RP2
.end
```
and the only CLI option I use in Python:
`cli = CliOptions(["./josim_minimal.cir"])`

The files I am using are packed here: [josim_minimal_filter.zip](https://github.com/user-attachments/files/16359581/josim_minimal_filter.zip)

- JoSIM netlist
- Jupyter notebook with analysis
- waveform.txt that contains the gaussian pulse
- minimal_a1.csv, minimal_v2.csv results from qucs-s simulation (overlayed in the notebook)

alicenprzemek, 2024-07-26:
Hello again,
I see now that this problem exists only for longer time steps. As I decrease the time step I see that there is less "energy leaked" and at around 1ps (instead of 10ps) I see a solid agreement.
[image]
(though the result from qucs is at 10ps time step still)

JoeyDelp, 2024-07-26:
Hi, I’ll have a look and see whats causing this. It’s most likely that qucs dynamically adjusts the timestep to correct errors caused by larger timesteps. JoSIM only does this correction for JJs. Large timesteps could directly result in energy loss from the differentials in the capacitors and inductors.

JoeyDelp, 2024-07-30:
Having had a look this is most definitely due to the differentials in the capacitor and inductor. Larger timesteps result in larger error and as a result incorrect simulation results. 

The QUCS technical papers mention adjusting the timestep if the local truncation error is too high, in which case the time step is reduced. 

Implementing this in JoSIM will take some time.

alicenprzemek, 2024-07-31:
Hey, good to know that setting a small time step is enough to solve it. Thanks for looking into it.