JoeyDelp/JoSIM #95: Shorted transmission line causes bad_optional_access

alicenprzemek, 2024-07-24:
Hello,
I've tried the following netlist:
```
CC1 nodeinput midpoint 50f
LL1 nodeoutput midpoint 10n
TLine1 nodeinput gnd gnd gnd TD=1n Z0=50
TLine2 nodeoutput gnd gnd gnd TD=1n Z0=50
VIN nodeinput nodeinputR cus(waveform.txt 10p 1.0 2)
RP1 gnd nodeinputR  50
RP2 gnd nodeoutput  50
.tran 10p 25000p 
.print NODEV nodeinput
.print DEVI RP1
.print NODEV nodeoutput
.print DEVI RP2
.end
```
where the transmission lines Line1 and Lin2 are shorted at one end. It leads to the following error:
`RuntimeError: bad optional access`
Adding a small resistor at the shorted node is a working hack, though I figured it would be good to let you know.

JoeyDelp, 2024-07-26:
Hi, thanks for bringing this to my attention. 
It seems I just never anticipated a transmission line being shorted and therefore never added the necessary checks for a case where both the input or output nodes are grounded.

JoeyDelp, 2024-07-30:
I have fixed this issue and pushed changes to the repository. Binaries will be compiled upon the next major release.

closed 2024-07-30 by JoeyDelp