JoeyDelp/JoSIM #105: Transmission line element 'T' does not apply delay (TD parameter has no effect)

Nishizacky, 2025-07-25:
It seems that the transmission line element `T` does not apply any delay to the signal, regardless of the specified `TD` (delay time) parameter. This severely limits its usability in high-frequency circuit simulations.

## Details
I have attached two files:
- One using the `T` element (named `_T.txt`)
- One using an equivalent LC ladder circuit (named `_LC.txt`)

[transmissionline_T.txt](https://github.com/user-attachments/files/21430197/transmissionline_T.txt)

[transmissionline_LC.txt](https://github.com/user-attachments/files/21430193/transmissionline_LC.txt)

In the LC-based circuit, the delay is correctly reproduced. However, in the circuit using the `T` element, the output waveform does not change at all even when the `TD` value is varied.

This suggests that the `T` element is not functioning as intended.

Nishizacky, 2025-07-25:
The screenshots I uploaded show the simulation results obtained by running the netlist files I previously shared.

As you can see, the circuit using the transmission line element `T` shows no delay, regardless of the `TD` value, while the LC ladder version behaves correctly.

[image]
[image]

JoeyDelp, 2025-07-28:
Looking at the component description and the syntax guideline for a transmission line (https://joeydelp.github.io/JoSIM/comp_stamps/#transmission-line)
what you are effectively doing here `T00                2   3   0   0              TD=10p               Z0=50` is grounding both the output nodes of the transmission line. 

Changing this to
`T00                2   0   3   0              TD=10p               Z0=50`
shows the 10ps delay you expect.

[image]

closed 2025-07-28 by JoeyDelp

Nishizacky, 2025-07-28:
I just re-ran the simulation with the corrected node connections, and I can confirm that the transmission line now functions correctly with the expected 10ps delay.

Thank you once again for your guidance and support!